MyRobotLab/InMoov #120: Utilities.aiml bug

hairygael, 2017-11-08:
Hello Anthony,
I discovered a bug and I think it lays in utilities.aiml
When you ask the robot to count to 5 for exemple, it will start to count until infinity and never can give the answer. MRL is then frozen counting endlesly.
```
<category><pattern>COUNT TO <set>number</set></pattern>
<template><think>
        <set name="count">0</set>
      </think>
      <condition name="count">
        <li>
          <value>
            <star/>
          </value>
        </li>
        <li>
          <set name="count">
            <map>
              <name>successor</name>
              <get name="count"/>
            </map>
          </set>
          <loop/>
        </li>
      </condition></template>
</category>
```

If you ask him to count to 10, it also doesn't respond the correct answer. Although with version 2563 was working. That is the version that I was using in Netherland and it was ok.
That one is in reductions2.aiml
I suspect for this one that you will say that it is ask.pannous that is borked...
```
<category><pattern>CAN YOU COUNT</pattern>
<template><srai>COUNT TO 10</srai></template>
```
If you have an idea where could the problem reside it would be great !

hairygael, 2017-11-08:
Update after a restart of MRL: count to 10 works again???? without changing anything... OKAY..

Count to 3 still throws MRL into space time.

moz4r, 2017-11-09:
Hi !
very strange issue... Fixed ( fun is I'm not really sure about what was the fix ... )
I suspect a space somewhere... 
Hoo we need to work on something to simplify aiml edition. 

Some ideas

- Add an new interpreter level to digest things :

```
    <category><pattern>DÉCONNECTE LE BRAS DROIT</pattern><template>Déconnection de mon bras droit.<oob><mrl><service>i01.rightArm</service><method>disable</method></mrl></oob></template></category>
```

CAN BECAME

`DÉCONNECTE LE BRAS DROIT | Déconnection de mon bras droit. | @i01.rightArm#disable`

- Better : a graphical thing with auto completion to easy find/edit something in the whole set ( I know someone who have some good knowledges about gui... )

...

Fix :
```
<category>
<pattern>COUNT TO <set>number</set></pattern>
<template>
  <think><set name="count">0</set></think>
  <condition name="count">
    <li><value><star/></value></li>
    <li>
      <set name="count">
        <map><name>successor</name><get name="count"/></map>
      </set>
      <loop/>
    </li>
  </condition>
</template>
</category>
```

hairygael, 2017-11-09:
Great it's fixed!!
Yeah a gui to edit the aiml would be the paradis!
The real issue is when you need to search in each file for a certain pattern, it takes me ages.

closed 2017-11-09 by hairygael