MyRobotLab/InMoov #121: face tracking +facerecognizer+people

hairygael, 2017-11-09:
Hello Anthony,
This must be very fun when using a powerfull PC!!!
Unfortunately the Lenovo tablet seems to drop the fps down to 2 frames per second and the head is moving way too slow to be able to track anything.
Do you have the same issue or is it really the tablet?

With that in mind:
I think we create a new voice command "Face detection" only for tracking faces(face detect filter), so my tablet can still enjoy searching faces for many years to come. :)  Many people bought Lenovo tablets because of my recommendation...
Voice command "Track" should remain to track a point.(LKTracking filter)
Voice command "Face recognizer" should remain for the face recognizer.(Face recognizer filter)

And we we use voice command called "Search Humans" for the face tracking+face recognizer+people.

:)

moz4r, 2017-11-12:
yeh power is fun ! I have decent results, not lot off fps but worky ( core i3 2ghz/ 4gb ram ).
I will change Voice commands based on your recommendation

I finish some magic before...

[image]

hairygael, 2017-11-12:
Hello Anthony,

What kind of super magic is that?!
Mode hors connexion avec utilisation de ce qui a déjà été downloadé? ou
alors avec sans utilisation de requête internet?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-11-13:
Offline mode using android offline voice recognizer capabilities. worky on my phone ( airplane mode to be sure ) need to test some others, offline capabilities can be different

moz4r, 2017-11-16:
Hi ! about your initial question, I remember now, pushed this thing some time ago.
It is enough or we need to change chatbot commands : 

https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/config/service_D_OpenCv.config.default#L5

hairygael, 2017-11-16:
> config/service_D_OpenCv.config.default#L5

Hello Anthony,
I don't have the time to try but it was a good option if the PC is strong.
:)
Can we change the chatbot command otherwise it means for to do face
recognition you need to shut down MRL, reconfigure and restart, not really
good for demonstrating.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-11-17:
pushed

https://github.com/MyRobotLab/inmoov/blob/f93eb4df8f68acdb6af7488addba860be1282504/InMoov/chatbot/bots/en/aiml/_inmoovGestures.aiml#L249

https://github.com/MyRobotLab/inmoov/blob/f93eb4df8f68acdb6af7488addba860be1282504/InMoov/gestures/trackHumans.py

closed 2017-12-05 by moz4r