MyRobotLab/InMoov #127: Updating

hairygael, 2017-11-29:
Somebody is asking if there is a way to automaticaly update InMoov.
Maybe we should have in the main config an option for that?
I know in the BAT file there is an MRL update being launched...

Привет из России
Подскажите как обновить систему робота? это должно происходить автоматически каждый раз при загрузке, или же какую-то команду нужно выполнить обновление?

supertick, 2017-11-29:
There is a rather good updating system in MRL called the "Agent".
http://myrobotlab.org/content/agent-or-how-mrl-starts 

It has potential to manage multiple instances and multiple versions of MRL and even have backups to easily revert to a previous version.  

[image]

But, it needs a maven structured repo, and maven is one of the projects started in the next release. So, a batch file script might be a good intermediate fix.

And since we are on the subject of next release, I assume you **only use the latest now on develop**?  It's stabilized to a point were we will never talk about 119 (or any other intermediate version) again ? :) 

There has been a metric shit-tonne of work done to the Manticore release, and I have been delaying as long as possible to put more auto-elf testing in (http://myrobotlab.org/content/boldly-scrub-where-no-one-has-scrubbed). But now, features like these will drive us into the next release as the changes to support things (such as a maven repo) are so large it requires separation from the stable base.

moz4r, 2017-11-30:
Yeh... it smell stable more and more !

[image]

At this time just replace the jar by a new one. The launcher will do the job for you, if a new version is detected, subconscious voice will tell you to restart mrl for update. 
As Grog say, tomorow things will change, the agent will have some work to do.

hairygael, 2017-12-02:
Thanks guys!
Okay lets be patient for the next release.
Currently, I never do just replacing the myrobotlab.jar because it almost always leads to an error.
I just tried to see if I was getting more lucky with recents version 1.0.2607with InMoov_develop-->1.0.2643 gives an error and freeze the process.
AttributeError: 'NoneType' object has no attribute 'lang_shutDown'

Maybe someone using the InMoov_master embeded within MRL, would get more lucky than me.

supertick, 2017-12-02:
Sure .. its more problematic now - and the "clean" way is to do a full install into a new directory.  We can work on the much better process "after" the release.

So can the InMoov community make a list of necessary fixes we can complete ?  We closed 70+ tickets ! Yay !

It would be nice to have some structure to the process, such as Gael Moz4r Markus Mats Kwatters and a few others sign off ?

moz4r, 2018-09-18:
https://github.com/MyRobotLab/inmoov/issues/70

closed 2018-09-18 by moz4r