MyRobotLab/InMoov #130: Fil de discussion Chatbot - [français]

moz4r, 2017-12-05:


moz4r, 2017-12-05:
Schéma actuel : 
( pour le modifier, utiliser draw.io : https://www.draw.io/#HMyRobotLab%2Finmoov%2Fdevelop%2Ftools%2Fshared_diagrams_documentation%2Fbrain_knowledge_semantic.xml ) 

[image]

Sebchko, 2017-12-05:
Hello.
Je ne sais pas ce qui cloche mais, apparemment la recherche wiki a un soucis chez moi...
bref, un petit print screen vaux mieux qu'un long disours.

[image]

Alors docteur, c'est grave?

moz4r, 2017-12-05:
mmmm , pas grave ! wikidata a changé un petit peu ses méthodes pour l'api, on a corrigé très récemment . Tu as gagné une mise à jour de myrobotlab.jar ( gaffe ça va écraser tes aiml )

Sebchko, 2017-12-05:
Merci! Il a contracté une petite régression des scripts python au passage mais rien qu'une bonne petite piqûre de rappel n'a pu rétablir.
Il est maintenant au top pour continuer à apprendre

Sebchko, 2017-12-07:
Bon assez assez travaillé en sous-marin. 
Afin que les choses soient claires, j'ai découvert AIML en même temps qu'InMoov du coup il y a certainement mieux à faire que ce que je propose en terme d'optimisation. Je suis ouvert à toute vos suggestions!

Histoire que la chose puisse être discutée, voila ce que je suis en train de faire sur les AIML FR:

 - éventuellement ré-indentation et fix des fonctions (cf utilities)
 - nettoyage des fichiers 
      -ajouts d'accents, correction orthographe, grammaire, conjugaison, suppression de données 
       personnelles d'ancien botmaster,...
      - agrégation de catégories en utilisant # XXX # au lieu de plusieurs catégories avec *
 - classement par sujet/fonction en récupérant les catégories ad-hoc dans les fichiers alphabétiques. (ex: 
    genre, métier, couleur favorite,...)
      -->user-profil , bot-profile (à venir), gestion de liste( à venir)...pour le reste je n'y ai pas encore songé
 - création de sets lorsque c'est possible (nationalité, pays, genre,color...) pour fermer les possibilités 
   réponses à une question précises. 
       - voire même dans un second temps pour regrouper des catégories qui sont traitées de la même 
         manière (par ex les favoris , sur l'exemple de bot-profile du jeux d'AIML en anglais).
 - rédaction de nouvelles réponses et de nouvelles déclarations (cf infra)
- ajout de commandes en fonction de l'inspiration (confirmation extinction par ex).
- comparaison avec le chatbot en langue anglaise et intégration de fonctions (contact, rsvp,...)
 
Pour l'instant les changements les plus visibles ont été introduits par la création de user-profile qui reprend des caractéristiques que l'utilisateur déclare en répondant aux questions du bot. Une partie de son contenu a été pompé sur le fichier "profile" (que je compte bien à terme assécher) et à gauche à droite dans le jeux de fichiers disponibles.

 Le fichier est organisé +/- dans l'ordre dans lequel les questions apparaissent dans "questions.aiml"
Un espèce de sommaire se trouve en tête du fichier pour que les futurs psychologues pour robot puissent s'y retrouver à moindre frais. (ce qui est l'argument principal du classement par sujet selon moi)

A priori, les déclarations spontanées sont traitées dans la mesure où la variable peux être déclarée.
(Ces caractéristiques sont stockées dans le fichier "username".predicate)

Le fichier user-profile est flanqué de user-profile-srai qui reprend les déclarations de l'utilisateur pouvant être dirigée sur une déclarations de caractéristique.
Aussi, pour plus de clarté, le fichier that-questions à été formaté de la même manière que les deux nouveaux.

Je compte débuter le même processus sur les "préférences/caractéristique" du bot sur base d'un fichier "properties.txt"contenant plus de caractéristiques qu'actuellement. 
Bref, il y a encore de quoi s'amuser pendant plusieurs semaines (mois)!

Pour la suite voila quelques idées en vrac que j'ai envie de développer pour le chatbot de InMoov dans l'esprit assistant personnel (et si elles existent déjà tant mieux!) :

- (bien sur) à terme ajouter une couche de neurones artificiels en lien avec le chatbot. Je n'y connais encore rien du tout...du coup les avis et idées sont les bienvenues mais je me dis qu'un chatbot correcte pourrait servir de base à l'entraînement du réseau.

-récupérer les mails, donner l'expéditeur et le sujet; effacer/lire ceux que l'utilisateur désire.
- envoyer des mails, tweet,...
- gestion d'agendas, de rappel, alarme, etc (rappelle moi de sortir la pizza du four dans 10'; va chercher le bouquet de fleur quand ma femme rentre, note mon rendez-vous )
- pour le fun donner l'horoscope sur demande (apparemment cette fonction a un jour existé sur un des 
  bots sources)   
-récupérer la météo à plus d'un jour (c'est possible avec openweathermap mais comme le service est 
  encapsulé...je ne sais pas encore y faire grand chose.)
-donner l'heure sur un autre fuseau horaires (quelle heure est-il à NYC)
- et d'autre qui m'échappent maintenant
- et celles qui vous font envies

J'ai cru comprendre que la crew qui a débuté le nettoyage l'année dernière s'est essouflée... sauf contre-ordre lorsque je serai passé par la case présentation sur le forum, j'ai l'intention de lancer un appel à volontaires pour éditer du contenu.
En espérant avoir été clair,
Enjoy

Seb

Sebchko, 2017-12-07:
Bon assez assez travaillé en sous-marin. 
Afin que les choses soient claires, j'ai découvert AIML en même temps qu'InMoov du coup il y a certainement mieux à faire que ce que je propose en terme d'optimisation. Je suis ouvert à toute vos suggestions!

Histoire que la chose puisse être discutée, voila ce que je suis en train de faire sur les AIML FR:

 - éventuellement ré-indentations des fichiers et fix des fonctions (cf utilities)
 - nettoyage des fichiers 
      -ajouts d'accents, correction orthographe, grammaire, conjugaison, suppression de données 
       personnelles d'ancien botmaster,...
      - agrégation de catégories en utilisant # XXX # au lieu de plusieurs catégories avec *
 - classement par sujet/fonction en récupérant les catégories ad-hoc dans les fichiers alphabétiques. (ex: 
    genre, métier, couleur favorite,...)
      -->user-profil , bot-profile (à venir), gestion de listes ( à venir)...pour le reste je n'y ai pas encore 
           songé
 - création de sets lorsque c'est possible (nationalité, pays, genre,color...) pour fermer les possibilités 
   réponses à une question précise. 
       - voire même dans un second temps pour regrouper des catégories qui sont traitées de la même 
         manière (par ex les favoris , sur l'exemple de bot-profile du jeux d'AIML en anglais).
 - utilisation de <condition> pour spécifier une réponse différente pour certaines caractéristiques d'une même catégories. (je suis avocat -->bonjour maître)
 - rédaction de nouvelles réponses et de nouvelles déclarations (cf infra)
- ajout de commandes en fonction de l'inspiration (confirmation extinction par ex).
- comparaison avec le chatbot en langue anglaise et intégration de fonctions (contact, rsvp,...)
 
Pour l'instant les changements les plus visibles ont été introduits par la création de user-profile qui reprend des caractéristiques que l'utilisateur déclare en répondant aux questions du bot. Une partie de son contenu a été pompé sur le fichier "profile" (que je compte bien à terme assécher) et par-ci par-là dans le jeux de fichiers disponibles. Pour le reste, j'ai aussi ajouté un peu de contenu.

 Le fichier est organisé +/- dans l'ordre dans lequel les questions apparaissent dans "questions.aiml"
Un espèce de sommaire se trouve en tête du fichier pour que les futurs robot-psychologues puissent s'y retrouver à moindre frais. (ce qui est l'argument principal du classement par sujet selon moi)

A priori, les déclarations spontanées sont traitées dans la mesure où la variable peut être déclarée.
(Ces caractéristiques sont stockées dans le fichier "username".predicate)

Le fichier user-profile est flanqué de user-profile-srai qui reprend les déclarations de l'utilisateur pouvant être dirigées sur une déclaration de caractéristique.
Aussi, pour plus de clarté, le fichier that-questions a été formaté de la même manière que les deux nouveaux.

Je compte débuter le même processus sur les "préférences/caractéristiques" du bot sur base d'un fichier "properties.txt"contenant plus de caractéristiques qu'actuellement. 
Bref, il y a encore de quoi s'amuser pendant plusieurs semaines (mois)!

Pour la suite voila quelques idées en vrac que j'ai envie de développer pour le chatbot de InMoov dans l'esprit assistant personnel (et si elles existent déjà tant mieux!) :

- (bien sur) à terme ajouter une couche de neurones artificiels en lien avec le chatbot. Je n'y connais encore rien du tout...du coup les avis et idées sont les bienvenus mais je me dis qu'un chatbot correct pourrait servir de base à l'entraînement du réseau.

-récupérer les mails, donner l'expéditeur et le sujet; effacer/lire ceux que l'utilisateur désire.
- envoyer des mails, tweet,...
- gestion d'agendas, de rappel, alarme, etc (rappelle moi de sortir la pizza du four dans 10'; va chercher le bouquet de fleur quand ma femme rentre, note mon rendez-vous )
- pour le fun donner l'horoscope sur demande (apparemment cette fonction à un jour existé sur un des 
  bots sources)   
-récupérer la météo à plus d'un jour (c'est possible avec openweathermap mais comme le service est 
  encapsulé...je ne sais pas encore y faire grand chose.)
-donner l'heure sur un autre fuseau horaire (quelle heure est-il à NYC)
- et d'autre qui m'échappent maintenant
- et celles qui vous font envies

J'ai cru comprendre que la crew qui a débuté le nettoyage l'année dernière s'est essouflée... sauf contre-ordre lorsque je serai passé par la case présentation sur le forum, j'ai l'intention de lancer un appel à volontaires pour éditer du contenu.
En espérant avoir été clair,
Enjoy

Seb

moz4r, 2017-12-07:
Yo ! y a toujours mieux à faire,  le principal c'est de faire ! et pour un gars qui débarque tu assure grave.
Essouflement sur le contenu du chatbot car la pauvre crew a beaucoup trop de choses a regler dans d'autres services, c'est vraiment cool que tu sois la !
Bonne idée, pour toutes tes idées surtout "rappel, va chercher
le bouquet de fleur quand ma femme rentre" et les neurones, obscure domaine fascinant a explorer

Au plaisir

[image]

hairygael, 2017-12-09:
Salut Sébastien,
Grosses avancées en perspective!
Effectivement ce qui compte dans un premier temps c'est de nettoyer toutes
ces fonctions qui ne fonctionnent pas et qui renvoient systématiquement sur
les malheureuses réponses par défaut.
Le chatbot Français est construit différemment de la version Anglaise et
j'avoue ne pas avoir vraiment regardé.
Les tournures de phrases et les fautes de Français étaient trop nombreuses
et cela m'a repoussé.
Je suis bien content que toi, Sébastien Bonnifet, viennes y mettre ton nez!
D'autant que je sens que tu vas t'amuser et que tu vas nous dépasser en
deux temps trois mouvements.

-Une fonction qui serait intéressante mais qui demande une conjugaison des
différents bots, serait de pouvoir basculer vocalement d'une langue à
l'autre.
 Par exemple:
 Humain: Peut-tu parler en Anglais?
 Robot: Bien sur, souhaites-tu que je passe en version Anglaise ou
simplement que je traduise ce que tu me dis?
 Humain: Passe en version Anglaise
 Robot: Hello, is this better?
 Ou bien (qui est déjà faisable):
 Humain: Peut-tu parler en Anglais?
 Robot: Bien sur, souhaites-tu que je passe en version Anglaise ou
simplement que je traduise ce que tu me dis?
 Humain: Traduis
 Robot: D'accord.
 Humain: J'ai une voiture bleu
 Robot: I have a blue car

 Cela demande une configuration au niveau des fichiers de config afin de
déterminer en amont le type de langue et type de voix sélectionnées.
 On pourrait envisager deux ou trois versions de langues compatible.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Sebchko, 2017-12-09:
Hellow!
une petite update du pull request (il reste encore pas mal de taf...bref)

@hairygael j'ai un peu chipoté en rapport à ton envie de switch entre les langues et j'ai un embryon de solution qui fonctionne chez moi mais je doute que ça fonctionne avec toutes les options envisageables. 
Si tu veux y jeter un oeil c'est dans "patch-1" sur https://github.com/Sebchko/inmoov/tree/patch-1

Il y a 5 fichiers à changer (configparser, _inMoov.config, chatbot.py, translator.aiml et traduction.aiml)
Ma langue par défaut est le français du coup il y qques petites modifs à faire
1/ _InMoov.config ajouter  MyVoiceType2FR="lavoixfrdetonchoix" (j'ai mis mon fichier en ex)
2/ ChatBot.py     l59 supprimer "2EN"
                           l65 ajouter MyVoiceType -->MyVoiceType2FR
3/ConfigParser   l57 EN --> FR

Ensuite il suffit de dire (enfin chez moi...) SPEAK IN FRENCH et tadaaa
Pour revenir en Anglais PARLE EN FRANçAIS
En gros ça relance le chatbot en Anglais après avoir sauvé les fichier AIML, changé de voix et changé de langue...

hairygael, 2017-12-10:
Bon alors grosse confusion.
Je t'ai confondu avec notre webmaster qui s'appelle seb également.

Sebchko, 2017-12-23:
Hello! 
Je suis en train de tester le chatbot et j'ai des difficultés à actualiser les aiml après avoir fait des modifs. 
Lors d'un dernier build de MRL le bouton "kill aimlif" faisait le taf mais il a été modifié en "restart (w/o AIMLiF". Depuis lors, les aimlif sont recopiés dans les aiml du coup...ben ça marche plus.

[image]

Sebchko, 2017-12-26:
Salut salut, 
j'avais un peu cafouiller avant de poster mon dernier message (supprimé depuis) le restart w/o aimlif fonctionne au poil!

Je voulais soulever un point à discuter:

il reste des vieux srai d'un chat bot source du genre "APPJOIE" ou "APPMOQUERIE" , à priori il me semble que cela servait à animer un visage pour le chatbot... (cf apparence.aiml pour toute la série.)
bref en gros, je me demandais si il y avait des projets pour ces fonctions?

Il est probablement possible de renvoyer certaines "expressions" vers des gestures, existantes ou à créer.
Ça vous tente? ou est-ce qu'on abandonne ce genre de fonctions pour le moment?

moz4r, 2017-12-27:
coucou bonne idée ça serait fun de pouvoir gerer quelques emotions de base ! 
J'ajouterais aussi un de ces 4 une librairie d'effets vocaux ( rires, soupirs ... ) comme on avait dans acapela. mais marytts n'est pas encore interfacé comme les autres tts pour gerer les fichiers audio.

hairygael, 2017-12-27:
Hello,
Oui ce serait une bonne idée de renvoyer vers des gestures.
Par contre évitons d'utiliser celle existante, à moins qu'elles ne
correspondent vraiment.
Le mieux c'est de créer de nouvelles gestuelles vraiment adaptées, toujours
dans un but de rendre le robot humain.
Nous, humain, sommes très alerte en ce qui concerne les signes des
mouvements corporel, alors si ceux ci sont inadaptés,
on ne considère pas possible de converser correctement.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Bruno-F, 2018-01-02:
Salut à tous, 
Pour rebondir sur le post de Sebchko, je suis volontaire pour filer un coup de main en échange d’explications claires et limpides sur la tâche à effectuer.
J’ai besoin de mieux connaître AIML et aurai qq questions avant de pouvoir être productif. 
@+

hairygael, 2018-01-03:
Hello Tous,
Il faudrait se refaire une liste des expressions possible corporellement
sans ou presque l'apport de mimique du visage. Seule la bouche ou les yeux
pouvant bouger cela demande un travail d'acteur pour bien déterminer ce que
le corps essaye d'exprimer.
Évidemment on peut toujours ajouter un mot dans la gesture.py ou l'AIML qui
confirme l'expression.
C'est d'ailleurs ce qu'ils avaient déjà fait pour certaines expressions
dans l'AIML apparence.

Depuis longtemps je voudrais utiliser la kinect one pour capturer des
gestuelles afin de les transcrire sur InMoov. Je l'avais déjà fait avec la
xBox360 mais comme elle ne prenait pas les mains en compte cela réduisait
encore plus le moyen d'expression. Malheureusement la Kinect One ne peut
fonctionner sur OpenNi.
Avec Virtual InMoov nous pourrions l'utiliser si les vitesses étaient
similaire au réel.
Ce qui serait top ce serait de pouvoir bouger Virtual InMoov en rigging
sous Blender en ayant une gestuelle filmé en arrière plan afin de bien
comprendre les mouvements.

Bon en attendant on peu faire comme j'ai toujours fait, bouger chaque servo
manuellement jusqu’à l'obtention d'une gestuelle acceptable. Attention de
désactiver le mode sleep dans inmoovLife.config sinon vous allez perdre
votre gestuelle avant la fin de sa création car InMoov va se mettre en
veille et se positionner en "rest".

Bruno, pour les AIMLs je peux peut-être t'aider dans la limite de mes
facultés.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Sebchko, 2018-01-14:
Salut salut, 

désolé du long silence, j'avais d'autres affaires sur le feu.Bref...

@Bruno-F Je ne sais pas si tu à déjà regardé ce que tu pourrais faire pour améliorer le chatbot en français, qui peut  sembler obscur au premier abord mais tu verras ce n'est pas si compliqué que ça.

A mon avis le plus simple et le plus agréable au départ est d'ajouter des réponses ou d'étoffer celles qui existent déjà en corrigeant au passage les diverses erreurs (orthographe, conjugaison, indentation, ponctuation,...) que tu rencontres. 
Par exemple tu peux ajouter des blagues, des variantes aux réponses sur la couleur des cheveux,...
Si tu te sens chaud, car ça demande un peu de réflexion, il est maintenant possible de demander la météo par période de 3h. 
Du coup tu peux ajouter des requêtes du genre: "quel temps fera t-il demain après midi?". Le truc c'est qu'il faut faire un calcul en fonction du moment de la requête...et envoyer la période (en heure divisé par 3 en gros ) entre la requête et la prévision voulue.
Va voir comment ça fonctionne dans weather.aiml et si ça te parle...enjoy.

Tu peux aussi si cela t'amuse, simplement parler avec le bot et observer la réponse et sa construction  dans le terminal. Si certaines entrées renvoient vers la réponse "je suis encore en développement" ou ses variantes. Tu peux ajouter la catégorie qui te sembles nécessaire à l'emplacement opportun et construire la réponse attendue. Pour débuter tu peux les lister et me contacter, on essayera de faire ça ensemble. 

Passer son temps à passer en revue les fichiers pour les corriger  est sans doute le meilleur moyen pour en avoir assez vite plein le dos...mais bon faudra y passer.
Pour ce dossier par exemple, il est possible de supprimer pas mal d'entrée. Par exemple les différentes variantes d'une catégories pour capter l'orthographe approximative de certains qui ne sont plus vraiment nécessaires dans la mesure où la reconnaissance vocale fait le travail.
Aussi, il y une multitude d'entrée et de réponse qui ne servent- à mon avis - à rien car vraiment pourrie ou obsolète qui peuvent soit être simplement supprimées (genre: "connais tu netscape?" ... ) ou si tu peux y trouver un sens, les réécrire dans un français compréhensible. (sauf si je n'ai pas encore découvert comment une entrée en français à peu près correct peut se voir martyrisée au point de ressembler à ce charabia par l'algo... je suspecte plutôt un bon passage par un programme de traduction automatique. Moz si tu connais ce Graal, je suis tout ouï ) 

Histoire de pas surcharger ce fil, si tu le désires on peut communiquer par mail ou sur le forum, je me ferai un plaisir de te raconter les quelques trucs que j'ai appris en AIML et développer un peu plus. Dans ce cas je t'expliquerai aussi le chantier que je me suis assigné, et si ça te botte on peut travailler ensemble dessus. 
Bref fais signe, si tu veux plus de précision.

Enjoy et joyeux non anniversaire à tous.

ps: allez je vous le dis quand même, je me suis commandé une imprimante 3d c'est pourquoi j'ai délaisser InMoov pour m'initier à Blender, me goinfrer de tuto, et tutti quanti.

Bruno-F, 2018-01-19:
Hi,
OK suis prêt à entrer dans le dur, on gagnera en efficacité par tel.
Suis joignable au O6 (trente sept) O5 24 (quatre-vingt seize) 
Faudrait trouver le support idéal pour les (nombreuses) questions/réponses à venir...