MyRobotLab/InMoov #135: OpenWeatherMap

Sebchko, 2017-12-17:
hello!
Juste un mot pour savoir si le service fonctionne correctement chez vous depuis la modif... sur ma bécane, ça me renvoi une erreur de clé API malgré en avoir généré une nouvelle sur le site.

moz4r, 2017-12-17:
Coucou essaye de changer la ville par défaut dans le fichier de config

Sebchko, 2017-12-17:
ouais, j'ai essayé avec une autre ville.  A priori ça devrait fonctionner avec la ville par défaut non? 
[image]

moz4r, 2017-12-17:
tu utilise bien le dernier jar ?  il y a eu des modif dans le service ( 2675 ) .

Sebchko, 2017-12-17:
oui, j'ai installé la dernière version de MRL au départ du bat de InMoov hier, histoire d'être certain d'être à jour...
j'ai simplement remplacé le dossier InMoov pour tester la version "develop"...

moz4r, 2017-12-17:
ok on Va essayer de debugger ça ! tu peux essayer de tapper ça dans python déja : 

OpenWeatherMap.setApiKey("ta clé")
isTheSunShiny("bruxelles",0)

Sebchko, 2017-12-17:
nada avec trois clés différentes... essayes en une pour voir...
97231973eb9c4cf309bdfe8f72181f62

moz4r, 2017-12-17:
Marche po la mienne : 2c09ab18cde7058fcfe08e1817cbe6c7

"Activation of an API key for Free and Startup accounts takes 10 minutes. For other accounts it takes from 10 to 60 minutes.
You can generate as many API keys as needed for your subscription. We accumulate the total load from all of them. "

Sebchko, 2017-12-17:
wtf! ta clé fonctionne 
bon c'est bien ce que je pensais...OWM me troll :)
ok, je vais en générer (encore)  une autre et voir ce que ça donne. 

merci moz!

Sebchko, 2017-12-18:
Dis j'ai un doute... il semblerait que la requête : 

"api.openweathermap.org/data/2.5/forecast/daily?q={city name},{country code}[&...&...]&cnt={cnt}"

renvoie une réponse pour les comptes payants. C'est clairement plus simple de récupérer la météo pour les jours suivants de cette manière mais si j'ai bien compris la requête: 

"api.openweathermap.org/data/2.5/forecast?q={city name},{country code}"[&...&...]&cnt={cnt}"

renvoie pour cnt=0 une liste de trois heures en trois heures sur les 5 prochains jours. (idem que 40)
renvoie pour cnt=x {pour x:1à40} la météo et les prévisions des périodes {H,...,H+3(x-1)h,...,H+117h}

il y a aussi la requête : 

"api.openweathermap.org/data/2.5/weather?q="; ...

pour la météo actuelle

De plus, il semble qu'il soit nécessaire dans certains cas de déclarer le pays. Par exemple: Bruxelles cela renvoie la météo d'un bled au Canada (ça à l'air d'un chouette coin cela dit  cf infra :), Brussels renvoie le temps de la ville qu'on imagine...
(ça à l'air d'un chouette coin cela dit :)
[image]

Bref, si tu as des choses qui t'inspire plus sur le feu,  je m'y colle dès qu'on aura avancé dans l'AIML et que je me serai mis sérieusement à JAVA et Phyton.

moz4r, 2017-12-20:
Bruxelles a bien changé :) pour corser le tout ils ont placé le vrai Bruxelles , en France...
Du coup en ville par défaut tu dois utiliser : bruxelles,fr 

ça c'est l'url qui est actuellement implémenté ( pour la météo du lendemain , par exemple )

http://api.openweathermap.org/data/2.5/forecast/daily?lang=fr&q=bruxelles,fr&mode=xml&units=metric&appid=aabd6a78592b9278aed93a12f95f868f&cnt=2

Sebchko, 2017-12-20:
ouais je ne conteste pas qu'il fonctionne... je dis qu'il fonctionne avec une clé payante :)

moz4r, 2017-12-20:
étrange j ai une clé gratos. Je vais essayer d en créer une nouvelle avec un autre mail pour voir

Sebchko, 2017-12-20:
sérieux?!

moz4r, 2017-12-20:
lol , je t'ai retrouvé , Brussels,be

En effet je crois qu'on a pas les même résultats en fonction de nos clés grrrrr.
Ils ont surement changé les CGV depuis que j'ai créé mon compte.

Tu as cité plus haut cette url ( forecast sans le daily ) : http://api.openweathermap.org/data/2.5/forecast?q=brussels,be&mode=xml&units=metric&appid=3e15be2149598b92634ec6d877c710cd&cnt=**9**

je crois que c'est une super solution non ?Tu peux faire qq test ? je mettrais a jour l'url dans le service

j'ai vu ça aussi, mais bon si on arrive a s'en servir avec les nouvelles clés ça sera pas la peine :

For ( free open source software ) developers: we welcome free and open source software and are willing to help you. If you want to use OWM data in your free software application please register an API key and file a ticket describing your application and API key registered. OWM will **review your request lift access limits for your key if used in open source application.** .

Sebchko, 2017-12-20:
sure, je m'occupes de ça demain! 
J'avais remarqué l'option 'open source' aussi...je me demandais si tu en avais fait la demande, 
c'était même ma dernière hypothèse  pour expliquer la différence entre nos deux comptes OWM. Bref...
Enjoy.

Sebchko, 2017-12-21:
Yo!
Bon je me suis penché sur la question et en ai profité pour tenter de modifier 'OpenWeatherMap.java' en local et de me faire mon petit build et tout et tout. Mais pour une raison encore obscure pour moi ant ne désire pas compiler le build à partir des clones de "myrobotlab" et du "repo" au départ des branches "develop" et je la sèches un peu. (il me reste l'histoire de ivy à explorer mais comme le  build se fait sans sur les branches master...si tu as une piste je suis preneur)

Je tentais de tester ceci : [https://gist.github.com/Sebchko/da9c7bf2573477429290dce28b5ecb23]
en gros modifier l' apiForecast et réutiliser le apiBase pour la météo du moment juste pour le fun car ce n'est pas nécessaire. (à propos de l'apiBase, elle renvoie 4 temps différents pour Bruxelles, mais ça ne m'étonne qu'à moitié :) 

Enfin soit, pour notre affaire, je pense qu'on peut utiliser l'url :  
"api.openweathermap.org/data/2.5/forecast?q={city name},{country code}"[&...&...]&cnt={cnt}"
de la même manière que l'actuel, en modifiant quelques peu les données que l'on récupère sur JSON (cf le lien).
Du coup il faut sans doute un peu changer le script python et l'AIML, à première vue pour 
 -python def isTheSunShiny(townParam="town",day=0):  -->day=1
- aiml: quel temps dans x jours ---> envoyer 8x à python
- à priori on pourra implémenter des requêtes du style : quel temps demain matin, jeudi après midi,..

voila pour où j'en suis pour le moment... je continue volontiers demain.
A t'naise pour le java, on est pas à la pièce, par contre si tu sais m'aiguiller pour le build de myrobotlab...

moz4r, 2017-12-22:
coucou ouaip cela semble la bonne voie ! ( le lien n'est pas bon )
ça c'est cool pour la precision des prévisions..
Tu t'attaque à java, tu es chaud, c'est bien ! Le problème c'est que ça devient vite adictif :) 
Au début j'ai aussi un peu galérer avant de pouvoir compiler quoi que soit... logiquement Ant doit faire le job tout seul.

[image]

Sebchko, 2017-12-22:
j'ai avancé un peu pour le build, enfin la compilation s'est lancée mais ça à foiré sur la fin:
[image]

no worries pour l'addiction, ça fait longtemps que je voulais apprendre à coder et un bon projet ça motive!
pff je sais jms dans quel parenthèse le mettre...
[](https://gist.github.com/Sebchko/da9c7bf2573477429290dce28b5ecb23)

moz4r, 2017-12-22:
Vérifie bien si tu as fait un checkout de la branche develop ( dans la vue git ) voici mon build.xml tu dois avoir le même

[build.zip](https://github.com/MyRobotLab/inmoov/files/1581135/build.zip)

Sebchko, 2017-12-22:
ouais c'est bien le même.
Je ne sais pas trop si c'était la solution mais j'ai crée une "home" pour ivy...et hop
[image]
mais ça me semble trop beau pour être vrai.

par contre il n'aime pas du tout mes modifs sur le java...

moz4r, 2017-12-22:
C'est l'heure de se leveeeeer :) j'ai poussé qq modif rapides basé sur ce que tu as fais.
Il faut essayer de boucler les modif ce soir, avant de de faire la mise à jour finale. C'est celle qui sera intégrée définitivement à manticore, qui ne devrait pas tarder ...

Sebchko, 2017-12-22:
hellow, je viens de terminer! 
java est prêt aussi

Sebchko, 2017-12-22:
[javaaaaa](https://gist.github.com/Sebchko/8e9e6f1c1cadf2085f6b7e7580062931)

moz4r, 2017-12-22:
Ahahaha puree ça c de la syncro

Sebchko, 2017-12-23:
Allez maintenant on peut clôturer cette conversation! qui aura eu le mérite de m'activer les neurones :)

closed 2017-12-23 by Sebchko