MyRobotLab/InMoov #151: Globalisation

moz4r, 2018-02-27:
We want python script minimal as possible, and things that are not daily usual can be hard coded. 
Today inmoov language pack is parsed from python land, goal is to push inside java land. 
The question is what is the best way.

Language pack is used exemple to talk about startup action, or robot critical errors.
( and of course for chatbot, but this part need to stay in python land )

https://github.com/MyRobotLab/inmoov/tree/master/InMoov/system/languagePack

moz4r, 2018-03-04:
Table of Language Culture Names, Codes, and ISO Values Method : https://msdn.microsoft.com/en-us/library/ee825488(v=cs.20).aspx

kwatters, 2018-03-05:
I recommend using the 2 letter ISO-639-1 codes...  I don't trust microsoft for "standards"  :)

https://en.wikipedia.org/wiki/List_of_ISO_639-1_codes

moz4r, 2018-03-05:
ahaha you right... But we need to deal with things like this, other standards source ?
https://github.com/MyRobotLab/myrobotlab/blob/master/src/resource/WebGui/service/js/WebkitSpeechRecognitionGui.js

moz4r, 2018-05-31:
ohhh discovering Locale class :) seem powerful

moz4r, 2018-06-07:
seem we need only 1 thing now, it is loadLanguagePack() . Like loagGestures()

closed 2018-07-01 by moz4r