MyRobotLab/InMoov #154: Gestures

DSBRobotics, 2018-03-12:
Will someone please explain where responses are stored in MRL with gestures. Several double speak, saying two things at same time. Some don’t seem to work at all. Example is “ battery level” that says a number and a percentage. Really not sure what it says. 
Also still have issue with mouth movement not syncing with output?.
Running Manticore

Thanks for your help in advance

hairygael, 2018-03-12:
Hello,
Battery level command is located in the AIML line 757.
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/_inmoovGestures
but also as srai function in, line 287, 339, 563,, 757, 991, 1391, 1921,
2084, 2367, 3157, 3267, 4004, 4044, 4395, 4698, 4925, :
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/reductions1
And
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/reductions_update
And
line 46, 86
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/bot_profile

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

DSBRobotics, 2018-03-12:
Thanks Gael

Where are learned responses stored?

hairygael, 2018-03-12:
It uses the AIML to determine what to store:
 C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/_InMoovKnowledge
And sets the learn into:
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\config/default.predicates
or in
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\config/yourname.predicates

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

DSBRobotics, 2018-03-14:
Thanks for sharing, still not finding where responses are stored. Are you using 1.0.2693 and if so do all the gestures work? I only get some to work and several say two things at once. Totally restarted yesterday out of frustration but same output. Guess I need to learn coding next!

Thanks again for your hard work. 

Scott

hairygael, 2018-03-14:
Hello,
Yes I use the 1.0.2693 in my links.
Can't you find the directories I am pointing to?
C:\Myrobotlab\myrobotlab.1.0.2693\InMoov\chatbot\bots\en\aiml/

If you have installed MyRobotlab as explained in my tutorial, you should be
able to find them as explained above.
Tutorial <http://inmoov.fr/how-to-start-myrobotlab/>
Responses are stored in all the AIML files, which means thousands of
answers possible but the learned are stored in the config/predicates as
explained above.
Exemple of learn possibilities:

address:4314 main street, salt lake city
age:32
baby:tommy
bestfriend:franck

address:where
age:how many
baby:who
bestfriend:who
birthday:when
birthplace:where
boyfriend:who
brother:who
cat:what
city:which
country:which
county:which
customname:unknown
daughter:who
dialnumber:unknown
dog:who
email:what
employer:who
equalto:false
father:who
favoriteactor:who
favoriteactress:who
favoriteband:who
favoritebook:what
favoritecar:what
favoritecolor:what
favoritedrink:what
favoritefood:what
favoriteicecream:what
favoritemovie:what
favoritesong:what
favoritesport:what
favoritetvshow:what
firstname:what
friend:who
gender:unknown
girlfriend:who
grandfather:who
grandmother:who
hair:what
has:what
he:who
husband:who
isanumber:false
it:what
lastname:what
latitude:which
likes:what
location:where
longitute:which
meta:set
middlename:what
mother:who
name:Friend
nephew:who
newcontact:unknown
newpattern:NORESP
niece:who
orientation:unknown
phonenumber:unknown
province:which
religion:unknown
residence:where
road:which
she:who
sign:which
sister:who
son:who
state:which
status:what
street:which
there:where
them:who
topic:unknown
town:which
voicemailnumber:unknown
we:who
wife:who

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

DSBRobotics, 2018-03-14:
I have found the directories but like Scott.predicates is empty and others dot see to have saved responses in them. 
The other issue is many gestures not opening. Maybe he doesn’t like my southern accent!

Scott

hairygael, 2018-03-15:
Regarding the Scott.predicates, you can erase it and also erase all AIMLIF
files, one of them might contain something that blocks the process.
Do not worry they will automatically be regenerated when you launch the
START_INMOOV.bat

You can test the gestures via text writting in the chatbot tab. Make sure
you write or say precisely the same command as found in the
_inmoovGestures.AIML.
For exemple if the pattern is:
<pattern>DISCONNECT EYES</pattern>

Write or say: DISCONNECT EYES

Anything else will be catched by a random answer, for exemple if you say:

DISCONNECT your EYES, it will not work and the robot might tell you: I
ASKED ANOTHER ROBOT BUT HE DIDN'T KNOW.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>