MyRobotLab/InMoov #161: make it so serviceData.json detects changes in service metadata

kwatters, 2018-05-23:
When changing metadata on a service (dependencies) the serviceData.json file needs to be deleted and re-generated.  It would be nice if this was a bit more auto-magic so it's not as confusing for developers.

alternatively, just get rid of the service data all together and compute it dynamically on startup.  The only info that we would need to keep around is the status to determine if a service has been installed.

supertick, 2018-05-23:
you can't compute it dynamically on startup - (at least in the jar - unless you explode the whole thing like a war file) ... the rub is the names of the classes in that package.

There is no way to determine the classnames of a package when its in jar form, unless you scan through "all" files of the jar - (shocking isn't it ?).  So serviceData.json is maintained by a generator - if it can't find the file it makes it.

MRL looks to see if it can be loaded as a resource, if it can it will try to generate serviceData.json ...  when it generates it, the generator scans the directory of source files to get the initial list of service names ..

This potentially could be replace by maintaining a package-info.java ...  but I haven't crossed over into that system ..

moz4r, 2018-07-13:
more magic ! serviceData.json made me crazy sometime

kwatters, 2018-07-24:
ok. i've updated the pom to make sure it cleans up the serviceData.json from a previous build.  I think this will fix it.. but we will need to test.  The next time we change version numbers of something in the getMetaData for a service we'll need to verify it pulls down the correct jars when the build comes from the cookie oven.

supertick, 2018-07-26:
moving to test column