MyRobotLab/InMoov #165: Possibility de stop a running gesture

moz4r, 2018-06-06:
gestures pushed inside a thread ?
any other "easy" idea ?

supertick, 2018-06-06:
Its a multi-level rabbit hole ...
First level is Python/Jython - interrupting python
Second level (potentially) is Arduino - interrupting movement

So lets go with first level first .. Please first tell how/where in the Python or InMoov script is the gestures started ?

You can't know ending without the beginning ...

supertick, 2018-06-06:
Historically I think I tried to re-initialize the interpreter - that was bad idea.
Looking and testing I can see we create a new Java/Python Thread PIThread for "most" new scripts that are executed ..  

The executing Java/Python thread will block when executing a script

There is one "global" reference to the last thread created, for the last script run (all other references are lost).

"If" you have a reference to the PIThread calling interrupt is effectively doing a Ctrl-C signal to the script.

My proposal would be to maintain PIThread references in a map and when stop() is called - iterate through all alive threads and interrupt them - then clear the map.  Currently there is a stop() function but it (partially) destroys the interpreter 

This solution would just make stop() work correctly

it offers no solution for scripts that are setup to block (but this seems to be an edge case) 

The map will fill during normal operation of dead thread references - but the overhead seems nominal to the desired effect.

supertick, 2018-06-07:
please test - if its to satisfaction lets close it

moz4r, 2018-06-07:
You are just magic !
Welcome to the new stopGesture()

Possibility to catch the error ? by exemple a log.info(Python process : KeyboardInterrupt)

KeyboardInterrupt: interrupted sleep

	at org.python.modules.time.Time.sleep(Time.java:469)
	at sun.reflect.GeneratedMethodAccessor45.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:206)
	at org.python.core.PyObject.__call__(PyObject.java:480)
	at org.python.core.PyObject.__call__(PyObject.java:484)
	at org.python.pycode._pyx116.loop$1(<string>:6)
	at org.python.pycode._pyx116.call_function(<string>)
	at org.python.core.PyTableCode.call(PyTableCode.java:171)
	at org.python.core.PyBaseCode.call(PyBaseCode.java:125)
	at org.python.core.PyFunction.__call__(PyFunction.java:403)
	at org.python.pycode._pyx193.f$0(<string>:1)
	at org.python.pycode._pyx193.call_function(<string>)
	at org.python.core.PyTableCode.call(PyTableCode.java:171)
	at org.python.core.PyCode.call(PyCode.java:18)
	at org.python.core.Py.runCode(Py.java:1614)
	at org.python.core.Py.exec(Py.java:1658)
	at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:267)
	at org.myrobotlab.service.Python$PIThread.run(Python.java:159)
------

closed 2018-07-13 by moz4r