MyRobotLab/InMoov #167: New translation, azure support languages

henrikmyl, 2018-06-09:
Hello,

I like to make new translation (finnish), but that's was not so simple what i first imagine. I can translate aim- files etc. but how can add new support language in AZURE TRANSLATOR FILE?  I think, i can't add new language manually in that file? Azure api needs updating?

I found Italian translation roadmap, so that helps me, but still i need more information..

Thanks.

hairygael, 2018-06-09:
Hello,
AZURE TRANSLATOR FILE is not what you want to look for. (Anthony correct me if I am wrong)
This file links to a AZURE account which uses the Microsoft commercial feature to do the translation.
This feature will translate what is requested, but the bot doesn't have the chatbot function to respond in Finnish.
For ex:
you say: "translate in Finnish my car is red" 
robot say:"auto on punainen"

What needs to be translated is the EN chatbot folder which contains the AIML and more.
For that matter, I updated the inmoov2-develop to be worky for Finnish, but I also updated the zip file available on InMoov website because it needed to have the Italian update anyway.
The version on the InMoov site is running the Manticore 1.0.2693 and it is fully working.
The version inmoov2-develop is to be compatible with Nixie 1.1.xxxx which is a work in progress.

If you proceed with translating the AIML and more, you can push the to inmoov2-develop branch for the Next release of MRL.
But you also can test them with the Manticore version.
Hopefully this is clear enough.
:)

henrikmyl, 2018-06-09:
Okei, thanks for the additional information. Now I know how to go ahead. I will test my translations before push them develop branch.

henrikmyl, 2018-06-10:
The tests did not succeed because I did not get the VoiceRss API to work. I try apiKey1=[a00xxxxx], but that not work. 

How that apikey should be written in service_5_Mouth.config file? 

Windows 10 speech recognition does not support Finnish, so i can't use LocalSpeech.

Translated:

*generalAPPS.lang
*Errors.lang
*general.lang
*StartingActions.lang 

*_inmoovChatbot.aiml

hairygael, 2018-06-10:
Hello,
I do not have apiKey for VoiceRSS. So I cannot test.
Try to run this script in myrobotlab python tab.

speech = Runtime.start("speech", "VoiceRss")
speech.setKey("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx")
speech.setLanguage("fi-fi")
speech.setRate(0)
speech.speakBlocking("se toimii, kyllä uskon, että se tapahtuu")

Remember that VoiceRSS is a bit slow, but the good side is that it's
caching the sentence once downloaded.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

henrikmyl, 2018-06-10:
Hello,
That script work fine. 

When i save VoiceRSS api in service_5_Mouth.config file and press start_inmoov, Myrobotlab give error: VoiceRSS error ( api key ? ): 

----------------

My service_5_Mouth.config file: 

[TTS]

Speechengine=VoiceRss

; you can use : 
;MarySpeech : open source TTS : http://myrobotlab.org/service/MarySpeech
;LocalSpeech : Local operating system TTS : http://myrobotlab.org/service/LocalSpeech
;Polly : [NEED API KEY AccessKey (apiKey1) and SecretKey (apiKey2)] http://myrobotlab.org/service/Polly
;VoiceRss : [NEED API KEY (apiKey1)] Free cloud service : http://myrobotlab.org/service/VoiceRss
;IndianTts : [NEED API KEY (apiKey1) and userid (apiKey2)] Hindi support : http://myrobotlab.org/service/IndianTts

VoiceName=0

; MarySpeech voices - take HSMM ones - http://myrobotlab.org/content/marytts-multi-language-support
; LocalSpeech : use local windows voices/macOs ( 0,1,2 etc ) print mount.getVoices()
; amazon polly :  http://docs.aws.amazon.com/polly/latest/dg/voicelist.html

[API_KEY]
apiKey1=[a00*********************]
apiKey2=

hairygael, 2018-06-10:
Ok,
I generated a apiKey for VoiceRSS to test and it works on my side using the
simple script.

I found an error in the* 5_Mouth.py* , you need to modify:

if tmplanguage=="it":tmplanguage="fi-fi"

with:

if tmplanguage=="fi":tmplanguage="fi-fi"

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

2018-06-10 18:05 GMT+02:00 gael langevin <hairygael@gmail.com>:

hairygael, 2018-06-10:
> marytts-multi-language-support

Try to write your key like this:

[API_KEY]
apiKey1=a00*********************
apiKey2=

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

henrikmyl, 2018-06-10:
Thanks a lot, now works! I noticed that  5_Mouth.py lines 139 - 142 need also updated?

Current:

  if i01.RobotIsStarted:

    if 'oui ' in text or 'yes ' in text or ' oui' in text or 'ja ' in text or text=="yes" or text=="oui":Yes()
    if 'non ' in text or 'no ' in text or 'nicht ' in text or 'neen ' in text or text=="no" or text=="non":No()

After the update:

if i01.RobotIsStarted:

    if 'oui ' in text or 'yes ' in text or ' oui' in text or 'ja ' in text or text=="yes" or text=="kyllä":Yes()
    if 'non ' in text or 'no ' in text or 'nicht ' in text or 'neen ' in text or text=="no" or text=="ei":No()

hairygael, 2018-06-10:
Good!!
You are correct I will update that line as well.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2018-06-10:
Do you have the _inmoovChatbot.aiml translated, so I can test it?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

2018-06-10 19:31 GMT+02:00 gael langevin <hairygael@gmail.com>:

henrikmyl, 2018-06-10:
Almost, but that file still requires work. I added it to develop branch..

hairygael, 2018-06-10:
Argh, Anthony is renaming the develop-branch and moving directories around.
Can't find the file anymore.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-06-10:
here is the new place ( useful to share aiml mods only and update language stuff , inmoov2 not yet worky ... )
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov2/languagePack/fi-FI

hairygael, 2018-06-10:
> languagePack/fi-FI

Thanks Anthony.
Henrikmil, can you push again your translated _inmoovChatbot.aiml because
during renaming I think it got lost.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

henrikmyl, 2018-06-11:
Github environment is new to me, but maybe you can see now what i have done? I don't have write permission in https://github.com/MyRobotLab/inmoov/tree/develop/InMoov2/languagePack/fi-FI

So i edit that file in my own fork and make pull request. Maybe that is not the right way to act?

LunDev, 2018-06-11:
No, that's one of the best ways to do it!

As soon as you're done, submit a pull-request to get your changes merged back into the project. Then somebody will review your changes and either merge them or request changes. You might also be asked to rebase your commits, but let's leave that for when the time comes round : )

henrikmyl, 2018-06-12:
Hello,

LunDev thanks for the answer. 

Hairygael any ideas what caused problem? FIRSTINIT start, but when i will answer the question "Hello, This is my first initialisation. I need some informations, I am listening, what is my name ?"  Script will not proceed...

When i use english language script works. For some reason, it compares the given name to the contents of the learnf.aiml file.

Picture attached:

[image]

hairygael, 2018-06-12:
Hello,
The aiml files work as whole with other files.
Currently you are trying to run a single aiml which needs many other files
to work.
The "sets" and "config" files have a lot to do with the learning system
because it is where new information are stored.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

henrikmyl, 2018-06-12:
Okei, now i think i understand a bit about how this works :) Thanks for help.

moz4r, 2018-06-13:
Hi ! you can now get the first initialization presentations works , added also default answers , as a sample

https://github.com/MyRobotLab/inmoov/tree/develop/InMoov2/chatBot/bots/fi-FI/aiml

for live testing after editing an aiml, you can use this button, so no need to restart every time

[image]