MyRobotLab/InMoov #168: Rename inmoov1 to inmoov2

hairygael, 2018-06-10:
Mmmh,
This is troubling me, because I was using a lot the develop branch as reference, also to fix small things for the working Manticore version.
For exemple I want to push some Finnish fix up, but the chatbot directory is gone.
I know it will be back in ProgramAB AIML directory, but for now, it's a bit unconvenient.
Would it be better to have created a inmoov2-develop branch instead of renaming?

moz4r, 2018-06-10:
Sorry big changes in progress ! inmoov develop branch must follow the main develop build.
Few solutions, but I'm afraid about duplicates updates

Maybe we can push again InMoov folder ( version 1 ) ?
commit to master branch ?

moz4r, 2018-07-02:
https://github.com/MyRobotLab/myrobotlab/issues/229

closed 2018-07-02 by moz4r