MyRobotLab/InMoov #171: Gesture creator

hairygael, 2018-06-14:
An idea for the new gesture creator version?
If that could work simulteanously with VirtualInMoov or in Real mode...
https://youtu.be/ECWUPJ0oDaQ

supertick, 2018-07-01:
references - https://github.com/MyRobotLab/myrobotlab/issues/228

supertick, 2018-07-01:
I think this will need to be done in the myrobotlab repo - so moving all info there ...

supertick, 2018-07-01:
closing as duplicate - please follow -> https://github.com/MyRobotLab/myrobotlab/issues/228
since the changes will need to be done to that repo

closed 2018-07-01 by supertick

hairygael, 2018-07-02:
Some functions of the GestureCreator are broken, but I guess that will
require a refactor anyway.
What seems to be broken is the script import and export, which makes the
GestureCreator currently useless.
I see MaVo is still around MRL, that's great news!
Maybe he will have a look into that super service he made.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>