MyRobotLab/InMoov #172: default nixie install and start_inmoov.bat throws some python errors

kwatters, 2018-07-11:
On the latest build (nixie) I get the following exception when trying to launch START_INMOOV.bat.

16:28:01.209 [python.interpreter.2] ERROR c.myrobotlab.framework.Service - python error PyException - null Traceback (most recent call last):
  File "<string>", line 41, in <module>
  File "InMoov/system/InitCheckup.py", line 49, in <module>
    execfile(RuningFolder+'services/'+filename.encode('utf8'))
  File "C:/nixie/InMoov/services/5_Mouth.py", line 239, in <module>
    setRobotLanguage()
  File "C:/nixie/InMoov/services/5_Mouth.py", line 205, in setRobotLanguage
    errorSpokenFunc('Language')
  File "C:/nixie/InMoov//system/Errors.py", line 70, in errorSpokenFunc
    print errorSpokenAlert.decode('utf-8')
UnboundLocalError: local variable 'errorSpokenAlert' referenced before assignment

        at org.python.core.Py.UnboundLocalError(Py.java:295)
        at org.python.core.PyFrame.getlocal(PyFrame.java:240)
        at org.python.pycode._pyx33.errorSpokenFunc$1(C:/nixie/InMoov//system/Errors.py:71)
        at org.python.pycode._pyx33.call_function(C:/nixie/InMoov//system/Errors.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyBaseCode.call(PyBaseCode.java:308)
        at org.python.core.PyBaseCode.call(PyBaseCode.java:132)
        at org.python.core.PyFunction.__call__(PyFunction.java:413)
        at org.python.pycode._pyx48.setRobotLanguage$9(C:/nixie/InMoov/services/5_Mouth.py:207)
        at org.python.pycode._pyx48.call_function(C:/nixie/InMoov/services/5_Mouth.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyBaseCode.call(PyBaseCode.java:125)
        at org.python.core.PyFunction.__call__(PyFunction.java:403)
        at org.python.pycode._pyx48.f$0(C:/nixie/InMoov/services/5_Mouth.py:272)
        at org.python.pycode._pyx48.call_function(C:/nixie/InMoov/services/5_Mouth.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1614)
        at org.python.core.__builtin__.execfile_flags(__builtin__.java:535)
        at org.python.core.__builtin__.execfile(__builtin__.java:512)
        at org.python.core.__builtin__.execfile(__builtin__.java:543)
        at org.python.core.BuiltinFunctions.__call__(__builtin__.java:88)
        at org.python.core.PyObject.__call__(PyObject.java:484)
        at org.python.pycode._pyx4.f$0(InMoov/system/InitCheckup.py:124)
        at org.python.pycode._pyx4.call_function(InMoov/system/InitCheckup.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1614)
        at org.python.core.__builtin__.execfile_flags(__builtin__.java:535)
        at org.python.core.__builtin__.execfile(__builtin__.java:512)
        at org.python.core.__builtin__.execfile(__builtin__.java:543)
        at org.python.core.BuiltinFunctions.__call__(__builtin__.java:88)
        at org.python.core.PyObject.__call__(PyObject.java:484)
        at org.python.pycode._pyx2.f$0(<string>:76)
        at org.python.pycode._pyx2.call_function(<string>)
        at org.python.core.PyTableCode.call(PyTableCode.java:171)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1614)
        at org.python.core.Py.exec(Py.java:1658)
        at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:267)
        at org.myrobotlab.service.Python$PIThread.run(Python.java:159)

kwatters, 2018-07-11:
Here's a link to a no-worky that produced the error.  http://myrobotlab.org/myrobotlab_log/upload/kwatters/1531341050.myrobotlab.log

moz4r, 2018-07-11:
Hi kwatters ! lots of this should evolve but we will get it worky for Nixie. fix in progress

kwatters, 2018-07-11:
I am starting to test the default inmoov scripts in nixie..  the next thing I'm confused about is starting the virtual inmoov.  It doesn't seem to launch.. but that's a topic for another ticket.  

the above error is because there's no default value for "errorSpokenAlert", so if an unknown alert comes in, it blows up in Errors.py.

moz4r, 2018-07-13:
Worky !

kwatters, 2018-08-14:
This looks good, thanks moz4r!  I'm going to close this ticket.

closed 2018-08-14 by kwatters