MyRobotLab/InMoov #177: captureGesture borked

moz4r, 2018-09-12:
IllegalFormatConversionException, need to reformat

wilcovantoorn, 2018-09-15:
What was the problem Anthony?

hairygael, 2018-09-17:
https://github.com/MyRobotLab/myrobotlab/commit/a735bab

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-09-17:
oh the captureGesture() function is supposed to "photograph" whole inmoov skeleton, by generating a python code. this python code could not be generated due to a number format problem and a missing function in the gui .

hairygael, 2018-09-17:
@Anthony Gallot <moz4r@free.fr>
do you think we could make a python fix to add for Manticore?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2018-09-17:
Also for Nixie release, is it something do-able to add the servoVelocity
involved during the capture to generate the python?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-09-17:
ohh good idea ! lets do this

moz4r, 2018-09-17:
About python fix for manticore :

#head
print("i01.moveHead("+str(i01.head.neck.getPos())+","+str(i01.head.rothead.getPos())+","+str(i01.head.eyeX.getPos())+","+str(i01.head.eyeY.getPos())+","+str(i01.head.jaw.getPos())+","+str(i01.head.rollNeck.getPos())+")\n")

#arm 
print "i01.moveArm(\"left\","+str(i01.leftArm.bicep.getPos())+","+str(i01.leftArm.rotate.getPos())+","+str(i01.leftArm.shoulder.getPos())+","+str(i01.leftArm.omoplate.getPos())+")"

moz4r, 2018-09-17:
maybe we need to talk about how to refresh this page : 
http://inmoov.fr/how-to-create-gestures-for-inmoov/
ear.addcomands / chatbot gestures, can be means of clarifying / optimizing
was thinking off a new command, chatbot.addcommands() that can feed/mod aiml files

moz4r, 2018-09-17:
ohh there is also some tickets about gesture Creator / orchestrator.
About actual gestureCreator gui service related to inmoov : dead or alive ? 
the service is there and people try to use it. I vote to fix it, waiting other things.

wilcovantoorn, 2018-09-17:
Maybe fix it for nixie?

If captureGesture is worky in manticore it should do for now

LunDev, 2018-09-17:
> I vote to fix it, waiting other things.

*_rewrite_ There's too much bad stuff in the logic behind it imo.
The gui might be worth keeping, but the background logic will probably need a complete redesign.

If the plan was to get Nixie out anytime soon, this should probably happen after Nixie / not be a priority for Nixie.

Further reading:
#171 MyRobotLab/myrobotlab#228

wilcovantoorn, 2018-09-17:
Im sorry, i got the names wrong, wasnt thinking. It should be the version
after nixie.
We should maybe make a plan on how we want the creator to behave. But maybe
thats another voting, fix or redesing... if its not too hard to insert
stuff in the future, i would vote fix. If, without borking it isnt easy, we
should redesing.

hairygael, 2018-09-17:
I just created a python gesture called captureGesture.py for Manticore
which includes:

#head
print("i01.moveHead("+str(i01.head.neck.getPos())+","+str(i01.head.rothead.getPos())+","+str(i01.head.eyeX.getPos())+","+str(i01.head.eyeY.getPos())+","+str(i01.head.jaw.getPos())+","+str(i01.head.rollNeck.getPos())+")\n")

#arm
print
"i01.moveArm("left","+str(i01.leftArm.bicep.getPos())+","+str(i01.leftArm.rotate.getPos())+","+str(i01.leftArm.shoulder.getPos())+","+str(i01.leftArm.omoplate.getPos())+")"
But the arm returns an error as "left" not behing viable.
Could we print in untitled.1.py tab instead of the python tab?

Regarding the gestureCreator/Orchestrator, it would be nice if it can be
fixed, but I wonder if it shouldn't work in a different way (hard to
explain when writing)
The idea is, whatever name and number of servo you launch with runtime, if
you open servo creator it detects the servos you have launched and includes
the min and max values.
Therefore the gestureCreator can adapt if you build an InMoov or another
sort of robot with only three servo.
This would avoid having a special service called gestureCreator for InMoov
that needs to be kept updated.

This page will need to be updated:
http://inmoov.fr/how-to-create-gestures-for-inmoov/

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-09-17:
updated py code for manticore : https://github.com/MyRobotLab/pyrobotlab/blob/develop/home/moz4r/Test/captureGesture.py

I'm not sure if we can do it better with ui, few things was broken.
But maybe some stuff like create a file inside gestures folder is possible.

hairygael, 2018-09-18:
Well for the Manticore version it's fine for now.
:)
I updated the version4 to version5 on InMoov website. Added a gesture
called captureGesture.py and modified the gesture aiml file.
I also updated the tutorial regarding captureGesture.
http://inmoov.fr/how-to-create-gestures-for-inmoov/

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-09-18:
nice, worky==happy