MyRobotLab/InMoov #180: get yolo "gesture" worky

moz4r, 2018-10-03:
Seem it is related to a personal build, we need to use yolo opencv filter

kwatters, 2018-10-03:
What do you want the gesture to do?  add the yolo filter?  what else?

moz4r, 2018-10-03:
I think it is for "what do you see"

kwatters, 2018-10-03:
you want to be able to respond with the current objects that are seen?  
yolo publishes that it recognizes on each frame.
I think you'd want to be able to ask the yolo filter what the last recognized objects are..  yes?

moz4r, 2018-10-03:
you right, I think we have tools to play with it

kwatters, 2018-10-03:
yeah. the info should be available in the "lastResult" object on the OpenCVFilterYolo  class..

moz4r, 2018-10-03:
[image]

moz4r, 2018-12-13:
We are close.. classification publisher need to be converted to java land, because odd race conditions between python sleep / speech recognition …