MyRobotLab/InMoov #183: Little issue bothering me

hairygael, 2018-10-09:
@moz4r, would you look into my little script
Seems I can't find a solution to keep the neo running during the script launches other sub scripts.
The neopixel animation runs for a very short time, while I would like it to run until the robot tells what is the analyse result.
[](https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/gestures/darknet.py)

moz4r, 2018-10-09:
Inside the inmoov develop branch ( nixie ), can we mod all related things to use yolo filter ? I know you had performance issues with it, but it wont worky if user don't have a moded manticore.
Maybe results are better now for filter?
if no, maybe we need to push something related to your mod ?

hairygael, 2018-10-09:
I am a bit confused..
Wasn't the InMoov2 branch for to develop for Nixie?
I think it would be safe to keep a branch working for Manticore even if we
abandone it later. Don't you think?
Maybe we should create a inmoov-develop_Nixie branch cloned from the
inmoov-develop.
And rename inmoov-develop to  inmoov-develop_Manticore

But otherwise I agree, we should review all what's necessary to make the
new branch work with the Yolo filters in Opencv.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-10-10:
- Inmoov develop branch must follow develop branch of myrobotlab develop branch ( actually dedicated to nixie release, next is ogre ) because it is a dependency downloaded by inmoov service.
- Inmoov master branch is freezed with last myrobotlab release ( so, actually master branch is for  manticore )
- InMoov2 branch is wip to Inmoov2 service ( goal is to delete a maximum of python sub files, for ogre release ).

Not sure for a Manticore branch .. My opinion is this will be a mess with multiple versions of gestures/aiml to take care, we need to fix Nixie I think, to forgot Manticore.

I think it is also technically possible to push important fixes to the master branch ( manticore ), if needed

hairygael, 2018-10-14:
Okay totally understood.
So the best would be to puch to master inmoov on Manticore release, what I have on the InMoov site.
[www.inmoov.fr/wp-content/uploads/mrl/myrobotlab.1.0.2693.7.zip](url)
Can you do that?

And I stop to add things for Manticore on the inmoov develop branch, but instead we focuse on Nixie.

moz4r, 2018-10-15:
PR sent, I pushed the inmoov folder from the zip
When merged I will update inmoov dependencies. 

So, after that ,you can remove all from the published zip from the website, exept myrobotlab.jar and start_inmoov.bat