MyRobotLab/InMoov #186: Aiml inmoov files disclaimer

moz4r, 2018-11-05:
While I'm thinking of it, we need to move disclaimer, before the <aiml> tag inside every files, because programAB engine delete it ( before tag is ok ).
Because sometime it is useful to point licenses type / rights...

hairygael, 2018-11-07:
> While I'm thinking of it, we need to move disclaimer, before the tag

Yes the disclaimer is auto produced by programAB, so we could remove it. It
keeps the file a little lighter on Git.
Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-11-10:
I think it's important to indicate the license in each file

moz4r, 2018-11-21:
related : https://github.com/MyRobotLab/myrobotlab/issues/340

hairygael, 2018-11-22:
> related : MyRobotLab/myrobotlab#340

Maybe a misunderstanding because of my previous answer.
Yes, we need a disclaimer in each files.
We can use the auto produced disclaimer using ProgramAb for the default
disclaimer and add what is necessary in each file.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

kwatters, 2018-11-23:
We should make it so ProgramAB does not overwrite the AIML files.  That seems to me to be the underlying issue here.