MyRobotLab/InMoov #189: Use embeded loadGestures() for py files

moz4r, 2018-11-28:
Aarrg , too many race conditions with python thread or sleep() because exotic and custom loadGestures() was used.

> time to use java class for it

supertick, 2018-11-28:
what?

moz4r, 2018-11-28:
oh yes will try to explain :) 
for now I use it inside python land : https://github.com/MyRobotLab/inmoov/blob/master/InMoov/system/InitCheckup.py#L93

That is hell.
Sometime a sleep() inside python land block a java thread, like a publisher.

i01.loadGestures() is better to load py files.

kwatters, 2018-11-28:
gesture files should only define a single method..  there should be no other code executed in a gesture file.  Any sleep statement within a method that is not triggered does not run.  I don't understand the problem here...

moz4r, 2018-11-28:
Yes we are agree. 
Problem here was the method to import gestures, I used execFile from python land to import py files. That cause race conditions sometime... With i01.loadGestures() I have no more issues ( for now )