MyRobotLab/InMoov #197: LeapData NotSerializableException

nickschiffer, 2019-10-19:
Hello, I am attempting to control remotely my inmoov arm. So far the RemoteAdapter Seems to be working properly. However, when I attempt to add the Leapmotion service to the PC node, I get a LeapData NotSerializableException. Any suggestions regarding how to control my servos on the RPi remote via the Leapmotion would be greatly appreciated. Thanks.

`[RemoteAdapterB] [ERROR] RemoteAdapterB error NotSerializableException - org.myrobotlab.service.data.LeapData java.io.NotSerializableException: org.myrobotlab.service.data.LeapData
	at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1184)
	at java.io.ObjectOutputStream.defaultWriteFields(ObjectOutputStream.java:1548)
	at java.io.ObjectOutputStream.writeSerialData(ObjectOutputStream.java:1509)
	at java.io.ObjectOutputStream.writeOrdinaryObject(ObjectOutputStream.java:1432)
	at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1178)
	at java.io.ObjectOutputStream.writeArray(ObjectOutputStream.java:1378)
	at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1174)
	at java.io.ObjectOutputStream.defaultWriteFields(ObjectOutputStream.java:1548)
	at java.io.ObjectOutputStream.writeSerialData(ObjectOutputStream.java:1509)
	at java.io.ObjectOutputStream.writeOrdinaryObject(ObjectOutputStream.java:1432)
	at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1178)
	at java.io.ObjectOutputStream.writeObject(ObjectOutputStream.java:348)
	at org.myrobotlab.net.TcpThread.send(TcpThread.java:319)
	at org.myrobotlab.net.TcpServer.sendTcp(TcpServer.java:156)
	at org.myrobotlab.service.RemoteAdapter.sendRemoteTCP(RemoteAdapter.java:306)
	at org.myrobotlab.service.RemoteAdapter.sendRemote(RemoteAdapter.java:296)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1356)
	at org.myrobotlab.framework.Service.invoke(Service.java:1292)
	at org.myrobotlab.framework.Service.run(Service.java:1689)
	at java.lang.Thread.run(Thread.java:748)
[RemoteAdapterB] [ERROR] removing mrl://RemoteAdapterB/tcp://192.168.1.31:6767 from registry`

supertick, 2019-10-19:
LeapData NotSerializableException means its not serializable.
Binary serialization which RemoteAdapter uses wants LeapData and LeapHand etc. to implement the Serializable interface.

I've updated the two files  on my branch, you might want to try to do the same on your branch
https://github.com/MyRobotLab/myrobotlab/commit/c4804cc8c5c4e53a258f0a9e8f5a06054061d406 

Also a work around is probably possible too using the Python service LeapMotion callback and a http service api call
e.g.
def onLeapData(data):
   http.get("http://raspi:8888/api/service/i01.leftArm.shoulder/moveTo/" + data.someValue")

nickschiffer, 2019-10-20:
Thanks for the quick response. I'll try your suggestions. In the meantime I have used `usbip` to virtually share the USB device (LeapMotion Controller) to the Pi so it appears as native on the Pi. I am ideally trying to be able to use `rightArm.startLeapTracking()` as shown [here](http://myrobotlab.org/service/leapmotion).