MyRobotLab/InMoov #207: How to Setup Software+Hardware

rut00, 2023-12-18:
I want to make speech module up and running with this code. I am beginner in this field. Please guide me with the setup. 

PS: I have extracted the .jar file and server is running.

hairygael, 2023-12-18:
Hello,
Have you followed this tutorial already?  It explains how to set the
brain(chatBot), ear(webkitSpeechRecognition), and mouth(TTS).
https://inmoov.fr/inmoov2-in-nixie/

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

rut00, 2023-12-18:
> It explains how to set the brain(chatBot), ear(webkitSpeechRecognition), and mouth(TTS). 

The tutorial provided are for Virtual environment or Hardware? How do I configure with Arduino ?

hairygael, 2023-12-18:
Hello,
it is not stucked, it is retrieving/downloading all the necessary
dependencies.
You need to be patient, especially if your internet or PC is a bit slow.
The link explains how to configure the Arduino.
You need to upload the MRLComm.ino to your arduino provided once Myrobotlab
is installed.
resource/arduino/MrlComm/MrlComm.ino
Once MRL is installed, in runtime tab you can either run in virtual mode or
real hardware mode.
[image: Capture d’écran 2023-12-18 120722.png]

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

rut00, 2023-12-19:
Thank you for the instructions. I have successfully done setup of MyRobotLab. I now have the question, on how to check the Vision with the simulator? And for face recognition, how do I train model on custom pictures?

Thank you so much for your prompt answer.

hairygael, 2023-12-19:
The simulator is mainly used to test or create gestures on your robot with
the servoMixer service.
 [image: Capture d’écran 2023-12-19 115428.png]

You can of course start the vision located within the head service.
You need to start the opencv service.
You can then train the robot for face recognition by clicking the TRAIN
button.
The chatBot will request a name and then approval.
Then it will start capture and take a picture, then it will start training
the face recognizer.
[image: Capture d’écran 2023-12-19 115043.png]

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

rut00, 2023-12-19:
For face recognition, can I add my custom python code? If so, how do I do that?

hairygael, 2023-12-19:
Yes you can either paste your code in
data/InMoov2/InMoovCustom.py which will be read automatically when InMoov
starts.
Or you can add your code in resource/InMoov2/gestures. It is less
recommended to tweak files in the resource directory.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2023-12-19:
Note that many of the scripts located in resource/InMoov2/gestures are used
with vocal commands of the chatBot or the UI

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

rut00, 2023-12-20:
I have few questions:

Can I access camera feed using just the Python Script? Can you please guide me for the same?
How to continuously keep Camera feed on, so that it can recognize people continuously?
How do I enable Human Head Tracking?
For QnA, I want the question to be directly searched in GPT3, how can I do this?

Thank you for the help.

hairygael, 2023-12-22:
Here below I assume that you are using the InMoov service.
You can do it via python this way:
  i01_opencv.startCapture()
  i01_opencv.stopCapture()

To keep the face recognition on continuously:
  facerecognizer()
To stop it:
  stopFaceRecognizer()

You can enable the headTracking in three ways:
1-in python
  TrackingHumans()
To stop it:
  stopTrackHumans()
2-with voice command in ear, brain, mouth:
 SEARCH HUMANS
To stop it:
  STOP YOUR TRACKING
3-In the InMoov UI:
  head-->opencv start-->headTracking-->enable/disable

To get your questions directly searched in GPT3 without the chatBot
interfering:
-use the voice command in ear, brain, mouth:
  GPT COMMUNICATION
To stop it:
  STOP GPT

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>