MyRobotLab/InMoov #208: Run Custom Python Script

Harshit-rathod-2484, 2023-12-20:
How do I enable, Python 3.x support? I tried changing value in `resource\Py4j\py4j.yml` from `useBundledPython: false` to `useBundledPython: true` but it didn't work.

I also, tried changing in the MRL framework and rebuilding jar file, but the version still remains 2.7.2.

I want to introduce my own Python script that can recognize face with Python 3.x version and want to install other Python libraries. Kindly please guide me through it.

Thank you for your help.

hairygael, 2023-12-20:
Hello,I normally just start the py4j service via the runtime tab. But maybe you are doing something different.Can you post your inquiry to the myrobotlab discord server? The developers will guide you better than I.Join the myrobotlab Discord Server!discord.ggFrom my iPhone, Gael

supertick, 2023-12-21:
[image]

supertick, 2023-12-21:
This is 1.1.1299 - verified python3 works, it requires the Py4j service to run.
If you get a "communication error" - then it means you have ghost Py4j servers running, which must be killed.

supertick, 2023-12-21:
[image]

Harshit-rathod-2484, 2023-12-21:
Okay How I can my system python version is Myrobotlab ?

Harshit-rathod-2484, 2023-12-21:
Thank you @hairygael ,

> Can you post your inquiry to the myrobotlab discord server? 

Could you provide link or server name? I haven't received in this reply.

rut00, 2023-12-21:
> This is 1.1.1299 - verified python3 works, it requires the Py4j service to run. If you get a "communication error" - then it means you have ghost Py4j servers running, which must be killed.

How do I kill those ghost Py4j servers?

hairygael, 2023-12-22:
@Harshit-rathod-2484, here is the link to Myrobotlab discord server: [https://discord.gg/aUqcafAt](url)