MyRobotLab/InMoov #209: Inquiry regarding Nervo Board Controller

xiangon11, 2023-12-21:
I had ordered Nervo Board Controller from InMoov site, but the order has not been dispatched. I have had several email communication but I am left with no response.

The order details:
 > Order number: 12470
 > Order Date: December 18, 2023

Please look into it and see if you could help with the same.

Thank you.

hairygael, 2023-12-22:
Hello,
Your order has been dispatched.
Please check in the spam box or emails for the tracking number and link to
your invoice.
If you cannot find them let me know.
Note that I am doing the Nervoboards preparation and shipping only when I
have some free time off my real work, so a bit of patience is required.

Best regards
Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

xiangon11, 2023-12-26:
Hello, thanks for answer @hairygael ,

The current one which you dispatched from La Poste, they are not responding. Is their another board in ready state which can be dispatched? And can it be shipped through FedEx?